picocomputer/rp6502 #3: Include file "pico/scanvideo.h" is missing

markrvmurray, 2023-01-07:
Hi

Thanks for this project!

I'm trying to build your code. I'm going to modify it to drive the quadrature clocks E/Q for an MC6809 processor, and having the ability to in effect create devices in software is a massive plus.

The build fails because the include file "pico/scanvideo.h" is missing:

```
:
[ 99%] Building C object src/CMakeFiles/rp6502_vga.dir/vga/term.c.obj
/Users/markmurray/Documents/PICO/rp6502/src/vga/term.c:11:10: fatal error: pico/scanvideo.h: No such file or directory
   11 | #include "pico/scanvideo.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/rp6502_vga.dir/vga/term.c.obj] Error 1
make[1]: *** [src/CMakeFiles/rp6502_vga.dir/all] Error 2
make: *** [all] Error 2
```

I've dig around in your repo, the Pico SDK and the Pico extra materials, and its not anywhere there. Could you add it please, or perhaps show me the error of my ways? :-)

Thanks!

rumbledethumps, 2023-01-07:
Maybe your PICO_EXTRAS_PATH has typo? That file is from here:
https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_scanvideo_dpi/include/pico/scanvideo.h

markrvmurray, 2023-01-07:
Indeed, I had a typo. I'd managed to confuse pico-examples with pico-extras.

Sorry for the noise :-(

markrvmurray, 2023-01-07:
Nothing to see folks. Move on please.

closed 2023-01-07 by markrvmurray