picocomputer/rp6502 #29: openocd no longer includes picoprobe.cfg

jchidley, 2023-09-10:
The current [README.md](https://github.com/picocomputer/rp6502) references `picoprobe.cfg` which is consistent with the `rp6502_vga.uf2` firmware reporting as *USB Serial Device (COM9), Picoprobe*. However this interface is not longer present in openocd. The newest [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) firmware reports as a *CMSIS-DAP v2 Interface*.

https://forums.raspberrypi.com/viewtopic.php?t=348242

rumbledethumps, 2023-09-10:
Use OpenOCD 0.11 or a third Pi Pico. I wasn't able to get the new CMSIS-DAP code working.

jchidley, 2023-09-11:
Thanks for that.  This is my workaround.  

Out with the old (the up-to-date one!):

```bash
cd ~
rm -rf openocd/
sudo rm -rf /usr/local/share/openocd
```

In with the new:

```bash
git clone https://github.com/raspberrypi/openocd.git --branch rp2040
cd openocd/
git checkout 610f137
./bootstrap
./configure
make -j8 # I have 8 virtual cores on my laptop
sudo make install
```

Wire up correctly

VGA | RIA
--- | ---
34 ADC2 | SWDIO
GND | GND
29 GP22 | SWCLK

```bash
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
```

Pico VGA now connects to Pico RIA

rumbledethumps, 2023-10-26:
Since picoprobe is no longer maintained, and CMSIS comes with an RTOS, I removed all debugging. Chances are good I'll use the resources to improve the graphics fill rate and built-in cortex debugging. That means it may never have the resources to return. A third Pi Pico or Debug Probe is cheap and was needed for VGA development anyways.

Closing this since there are no plans to bring internal debugging back.

closed 2023-10-26 by rumbledethumps