picocomputer/rp6502 #34: 0.2 rp6502 vga build fails with RAM memory overflow

ulften, 2023-10-14:
Probably a beginners issue! I finally got down to trying to build the images for the picos for the first time. I used VS code. The code is taken from the 0.2 release. The ria built just fine (not tested executing), but the vga build fails with a RAM memory overflow check (~17+ kB too big). Checking the linker config file, it seems that is using all available ram on the pico. Since I am using Visual Studio, I guessed that maybe the Debug config added the extra space and tried to tried to build not using debug info, but failed.

My goal here is to start experimenting with a Swedish keyboard layout, so maybe I don't need to modify the vga image, but it would still be good to know how to build new images from my own code. Thanks, Ulf

rumbledethumps, 2023-10-14:
VGA needs a release build until I make a custom linker map. Totally normal for microcontroller development.

ulften, 2023-10-15:
Ok, thanks!

closed 2023-10-15 by ulften