picocomputer/rp6502 #55: How is other IO besides the VIA mapped?

cbmeeks, 2024-07-04:
I've been studying the schematic and, correct me if I'm wrong, it would seem the VIA would be turned on with the following addresses:

FF00 to FFE0.

It is only setting pins A15:A5 does the VIA turn off.  Meaning, if IORQ = ON and RREQ = OFF, then the VIA is enabled.

How would, for example, another VIA be selected with that memory map?

**VIA IS ON**

[image]

** VIA IS OFF**

[image]

Thanks

rumbledethumps, 2024-08-04:
What "other VIAs" are you trying to select? There's only one in the reference design.

cbmeeks, 2024-08-04:
I mean if I wanted to add another VIA or some other IC.

rumbledethumps, 2024-08-04:
There's only one VIA in the reference design. This is working as intended. If you want to design something else then you have to do the engineering.

closed 2024-08-04 by rumbledethumps

cbmeeks, 2024-08-04:
I think you are getting too hung up on "the other via".  My point was if I wanted to add other IC's to the design in addition to the existing via.  It doesn't seem possible with this design.

But ok....lol